GyverLibs/GyverStepper #3: Не работают примеры на основе keep_speed

demosspro, 2021-11-28:
Подключил
```
#include <GyverStepper.h>
GStepper<STEPPER4WIRE> stepper(2038, 7, 5, 6, 4);
// мотор с драйвером ULN2003
void setup() {
  stepper.setRunMode(KEEP_SPEED); // режим поддержания скорости
  stepper.setSpeedDeg(50);        // в градусах/сек
}

void loop() {
  stepper.tick();
}
```
и тишина - горят только светодиоде на драйвере - (out 3 и 4 )
-
и так далее на основе.
-
Примеры с keep_pos  работают.

GyverLibs, 2021-11-28:
привет, спасибо большое!! Ошибку нашёл, обновление выйдет сегодня ночью

closed 2021-11-28 by GyverLibs

demosspro, 2021-11-28:
Вам спасибо! За развитие сообщества любителей электроники!
Дай вам бог множества светлых дней за любимым делом!

demosspro, 2021-11-28:
> привет, спасибо большое!! Ошибку нашёл, обновление выйдет сегодня ночью

Вам спасибо! За развитие сообщества любителей электроники!
Дай вам бог множества светлых дней за любимым делом!