GyverLibs/GyverStepper #4: esp32 overload

iqubik, 2022-01-07:
esp32 examples in stepper2 can't be compile

In file included from C:\Program Files (x86)\FLProg738\ideV7\portable\sketchbook\libraries\GyverStepper\examples\Stepper2\sweep\sweep.ino:3:0:
C:\Program Files (x86)\FLProg738\ideV7\portable\sketchbook\libraries\GyverStepper\src/GyverStepper2.h:202:10: error: 'void GStepper2<_DRV, _TYPE>::setSpeed(int)' cannot be overloaded
     void setSpeed(int speed) {
          ^
C:\Program Files (x86)\FLProg738\ideV7\portable\sketchbook\libraries\GyverStepper\src/GyverStepper2.h:190:10: error: with 'bool GStepper2<_DRV, _TYPE>::setSpeed(int32_t)'
     bool setSpeed(int32_t speed) {
          ^
C:\Program Files (x86)\FLProg738\ideV7\portable\sketchbook\libraries\GyverStepper\src/GyverStepper2.h:347:10: error: 'void GStepper2<_DRV, _TYPE>::setMaxSpeed(int)' cannot be overloaded
     void setMaxSpeed(int speed) {
          ^
C:\Program Files (x86)\FLProg738\ideV7\portable\sketchbook\libraries\GyverStepper\src/GyverStepper2.h:333:10: error: with 'void GStepper2<_DRV, _TYPE>::setMaxSpeed(int32_t)'
     void setMaxSpeed(int32_t speed) {
          ^
exit status 1
Ошибка компиляции для платы ESP32 Dev Module.

iqubik, 2022-01-07:
#ifndef ESP32 вместо #ifndef ESP8266 в файле stepper2 дают скомпилировать под eps32
Но уходит совместимость с esp8266

AlexGyver, 2022-01-07:
Пофиксил в репо, релиз выйдет после тестов на железе

iqubik, 2022-01-15:
Спасибо!

closed 2022-01-15 by iqubik