GyverLibs/GyverStepper #7: Kernel panic Speed=0

iqubik, 2022-01-24:
На esp32 если вызвать функцию конструкцию типа:
  stepper.setMaxSpeed(0);     // скорость движения к цели
  stepper.setAcceleration(0); // ускорение
  stepper.setTarget(10000);       // цель
Получаем kernel panic и постоянную перезагрузку контроллера.
Можно ли сделать обработчик такого неверного запуска движения, который будет просто говорить Готово?
Спасибо!

GyverLibs, 2022-01-29:
Это stepper или stepped 2?

iqubik, 2022-02-06:
> Это stepper или stepped 2?
Stepper2

GyverLibs, 2022-02-08:
Пока не нашел причин для паники, но в ближайшем обновлении постараюсь поправить

GyverLibs, 2022-02-09:
не смог понять что не понравилось kernel'у, при задании нулевой скорости там сразу выход из функции без применения этой скорости, и скорость остаётся прежней. Сейчас сделал, чтобы задание нулевой скорости вообще не позволяло запустить setTarget. Жду тестов

GyverLibs, 2022-02-14:
Да, пофиксилось

closed 2022-02-14 by GyverLibs