GyverLibs/GyverStepper #8: Чуть больше документации к коду

zag2art, 2022-02-06:
Изучаю код планеров. Функции вполне нормально документированы, но чуток не хватает описания приватных полей. Приходится по всему коду бегать, чтобы собрать, что за что отвечает. Спасибо за модуль.

GyverLibs, 2022-02-06:
Приватных? Расписать за что отвечают приватные переменные?

zag2art, 2022-02-06:
Да, если не трудно. Разбираюсь как это все работает.

GyverLibs, 2022-02-06:
Не знаю насколько это поможет, я уже сам наверное с трудом разберусь как оно работает) очень сжатые пережатые алгоритмы

zag2art, 2022-02-06:
меня в основном планеры интересуют. сравниваю с тем как это сделано в grbl

GyverLibs, 2022-02-06:
А есть где почитать по grbl? Я в свое время не смог разобраться и решил что проще написать с нуля свой планнер

zag2art, 2022-02-07:
почитать я не нашел, но есть исходинки с самой первой версии. Вот, например, версия [(0.1)](https://github.com/grbl/grbl/commit/9df29ad3b3fb769317d9c5560c54e89c92a71680), я же смотрю версию [0.6](https://github.com/grbl/grbl/tree/v0_6) - последнюю, которую норвежец писал, там вроде и функционал уже основной и по структуре все устоялось.

Судя по всему в grbl похожий подход. Тоже строят трапеции по скорости, стараются несколько трапеций плавно соединить, чтобы избежать рывков, ну и интервалы между степами контролят по теймеру через прерывания.

zag2art, 2022-02-07:
https://awesome.tech/grbl-demystified/
https://openbuilds.com/threads/ask-grbl-cnc-questions-here.3000/
Вот все, что нашел по докам

GyverLibs, 2022-02-07:
У меня тики как раз чтобы тикать в прерывании таймера :) Просто это полностью вынесено в скетч, чтобы можно было завести на любой платформе, там есть для примера как на 328 Меге работать с таймером.

У грбл используется brezenham? Я не придумал как сделать лучше, но с ним все остальные моторы (кроме нулевого) движутся неравномерно и вибрируют больше чем нужно

zag2art, 2022-02-07:
> У меня тики как раз чтобы тикать в прерывании таймера :) Просто это полностью вынесено в скетч, чтобы можно было завести на любой платформе, там есть для примера как на 328 Меге работать с таймером.

Все верно, нужно абстрагироваться от железа. Если бы в grbl использовали этот же подход - код был бы во много раз проще. Там же все в кучу намешано.

> У грбл используется brezenham? Я не придумал как сделать лучше, но с ним все остальные моторы (кроме нулевого) движутся неравномерно и вибрируют больше чем нужно

Разбираюсь с этим. Как минимум, при рисовании линий в grbl они тоже находят ось, по которой дельта максимальна и смотрят сколько это степов. Именно это число и будет step_event_count - то есть числом тиков, по которым будут толкаться степперы. Так что похоже, что без Брезенхема тут никак.

zag2art, 2022-02-07:
Нашел, кстати по Брезенхему и как они борятся с вибрацией - https://github.com/grbl/grbl/blob/master/grbl/stepper.c#L233

GyverLibs, 2022-02-07:
прикольно, надо над этим подумать! Но вибрации там не из за этого, а из за того, что мотор не имеет постоянной скорости. Основной мотор шагает с чётким периодом, а вот остальные могут делать условно два шага через шаг, то есть "шаг-шаг-пропуск-шаг-шаг-пропуск..." и в таком роде. Если увеличить разрешение - походу можно этого избежать, надо попробовать в симуляторе

zag2art, 2022-02-07:
Да, похоже, что с увеличением разрешения можно сделать пульсы второстепенных моторов более равномерными. Вместо "110110" будет что-то типа "100100100100" - соответственно и звук будет больше на синусоиду похож.

GyverLibs, 2022-02-07:
Ух попробую как будет минутка)

GyverLibs, 2022-02-07:
если интересно, вот так у меня реализован брезенхем, код можно запустить в Processing. Тут на 2 оси, названия переменных +- такие же как в гуверпланере
```cpp
int scale = 5;
int dS[] = {0, 0};
int tar[] = {0, 0};
int pos[] = {0, 0};
int nd[] = {0, 0};
int steps[] = {0, 0};
int dir[] = {0, 0};
int S, maxAx, step;

void setup() {
  size(800, 800);
  noStroke();
  fill(0);
}

void draw() {
  background(255);
  tar[0] = mouseX / scale;
  tar[1] = mouseY / scale;
  pos[0] = width / 2 / scale;
  pos[1] = height / 2 / scale;
  steps[0] = pos[0];
  steps[1] = pos[1];
  
  S = 0;
  for (int i = 0; i < 2; i++) {
    dS[i] = abs(tar[i] - pos[i]);
    dir[i] = pos[i] < tar[i] ? 1 : -1;
    if (dS[i] > S) {
      S = dS[i];
      maxAx = i;
    }
  }
  for (int i = 0; i < 2; i++) nd[i] = S / 2;
  step = 0;

  while (step < S) {
    step++;
    square(steps[0] * scale, steps[1] * scale, scale);
    for (int i = 0; i < 2; i++) {
      if (i == maxAx) steps[i] += dir[i];
      else {
        nd[i] -= dS[i];
        if (nd[i] < 0) {
          nd[i] += S;
          steps[i] += dir[i];
        }
      }
    }
  }
}
```

GyverLibs, 2022-02-07:
собственно осталось домножить переменную брезенхема и общее расстояние на множитель, и получить дробные шаги!
```cpp
  for (int i = 0; i < 2; i++) nd[i] *= mult;  
  S *= mult;
```
[image]

zag2art, 2022-02-07:
Красота, видно, что оба движка толкаются равномерно. Как я понимаю, это при делении на 8 шагов. Адаптивный вариант будет посложнее. Но если железо держит - вариант отличный.

zag2art, 2022-02-07:
Поставил себе processing - удобная штука, надо будет ваши видео по нему глянуть.

GyverLibs, 2022-02-07:
Адаптивный - просто выбирать делитель в зависимости от времени usMin, и всё. Прикинуть, какая частота будет критической и не превышать её. Осталось придумать, как работать с повышенным разрешением брезенхема в условиях планировщика, возможно нужно пропускать n шагов для пересчёта времени. Там надо подумать, как нибудь на неделе займусь

zag2art, 2022-02-08:
Будете дорабатывать, добавьте, плиз чуток комментов к приватным переменным

zag2art, 2022-02-09:
Кстати, даже если mult будет равен 8, то тиков все-равно будет максимум 3 - по количество моторчиков. Один основной, который по главной оси, и от 0 до 2-х второстепенных. Получается, какой бы mult не был (хоть 32, хоть 64)- это просто точность повышает timestamp-а тиков дополнительных степперов. количество же событий не увеличивается.

AlexGyver, 2022-02-09:
именно поэтому нужно делать "псевдотикер" внутри основного с пропуском sub-шагов

GyverLibs, 2022-02-09:
Так, доделал ролик на завтра, сел попробовать степпер. Заработало сразу, вот до и после
[image]
[image]

zag2art, 2022-02-10:
И как звук с вибрациями?

GyverLibs, 2022-02-10:
Да, гораздо лучше

closed 2022-02-10 by zag2art

zag2art, 2022-02-10:
Я уже понял, что это о равноускоренном движении. [тут](https://physics.ru/courses/op25part1/content/chapter1/section/paragraph4/theory.html#.YgVjYO5Bz-Q), например, все есть.