GyverLibs/GyverStepper #14: stepper.setAcceleration на разных stepper.setMaxSpeed живет своей непонятной и загадочной жизнью

Brandvik, 2022-12-17:
Всем здравствуйте. 
Библиотека 2.6.4
328Р
Загружаем пример из Stepper2 - StepperControl 
задаем 
stepper.setAcceleration (100000)  // на 100 000 особенно почему-то все ломается , на 50 000 и ниже, работает сносно и хорошо stepper.setMaxSpeed (5000)          //  а так же 10000, 15000 и 20000
 
int16_t path[] = {0, 16000};   

Пробуем в плоттере (в жизни все так же) остановить ,функцией  stepper.stop()  (S), мотор на разных этапах движения... 
Наслаждаемся странным поведеним торможения
Ускорение после 50 000  быстрее не становистя, с делителем 16 очень грустно все разгоняется...
Потом попробуйте ввести 
stepper.setAcceleration (200000)
Получаем по факту как будто ввели 2000... 
 В общем что это все, баг или фича ?

GyverLibs, 2022-12-17:
Чувствую что с такими огромными числами происходит переполнение 32 бит при вычислениях

GyverLibs, 2022-12-17:
Ну и стоило бы с документацией ознакомиться
```cpp
void setAcceleration(uint16_t acc);
```

Brandvik, 2022-12-17:
От я "Маша"!  И ведь читал-перечитывал прежде чем ту написать. А на тип данных не обратил внимание.... Так с этим разобрался. 
Но можно ли как-то увеличить ускорение разгона торможения? Как я уже говорил, при работе с делителем 16 ускорение может и должно быть быстрее, а то кокой то вялый огурец получается
И остается вопрос, почему торможение при вызове функции стоп происходит медленнее чем разгон\торможене при штатной работе функции  setTarget ? Особенно это заметно когда стоп вызван чуть раньше чем начинается "штатное" замедление, скорость падает и мотор доходит до заданного значения, а скорость еще не 0 и происходит резкая остановка. Вроде шаги не теряются, все таки скорость близка к нулю, но как то "не приятно" слышать "удар".

GyverLibs, 2022-12-17:
> можно ли как-то увеличить ускорение разгона торможения?

попробую в следующий раз как буду ковырять библиотеку

> торможение при вызове функции стоп происходит медленнее чем разгон\торможене

на каких настройках скорости/ускорения? Работа через прерывания или просто в лупе?

Brandvik, 2022-12-17:
`на каких настройках скорости/ускорения? Работа через прерывания или просто в лупе?`

 работа в лупе, но по сути голый тэстовый скетч опрос двух ног для кнопок да асинх. вывод позиции в терминал 
stepper.setAcceleration(50000);
  stepper.setMaxSpeed(20000);

Выглядит так

[image]

А вот "ударная" остановка
[image]

На скорости 10000 это тоже все проявляется

Со снижением ускорения в два раза эффект не проявляется или просто на глаз становится не заметен...

GyverLibs, 2022-12-17:
записал себе. Где то в вычислениях траектории возможно что то переполняется и считается неправильно

Brandvik, 2022-12-18:
Не знаю, актуально ли будет, но вот еще информация, может окажется полезной.
Завел на прерывания и получил такую картину, после полного разгона по функции стоп  торможение срабатывает строго через раз в любом месте траектории...
 
[image]

Когда тикало в лупе такого эффекта не наблюдалось.
ЗЫ Если не в лом. ткните носом почему на таймерах это работает

```
stepper.setTarget (6400);
  setPeriod(stepper.getPeriod());
    startTimer();
```
  а это не работает   
```
 stepper.setSpeed(-10);
    setPeriod(stepper.getPeriod());
    startTimer();
```

 и вобще не работает setSpeed если НЕ тикать в лупе?

А если завести таймер и тикать в лупе то скорости setTarget  в два раза вырастают :) Но это вопрос риторический...

GyverLibs, 2022-12-18:
Вот тут видимо нужно добавлять volatile. Странно что раньше не всплыло, я очень плотно тестировал и с прерываниями и без

GyverLibs, 2023-07-18:
В текущем обновлении нашел и исправил проблему со слишком медленной остановкой при высоких ускорениях

closed 2024-01-25 by GyverLibs