GyverLibs/GyverStepper #15: GyverStepper2 Ошибка компиляции при выборе setMaxSpeedDeg

Roman-RV78, 2022-12-18:
Здравствуйте. При выборе функции setMaxSpeedDeg вместо setMaxSpeed, ошибка компиляции

_D:\Temp\.arduinoIDE-unsaved20221118-6196-5kjgm6.x7t35\sweep\sweep.ino: In function 'void setup()':
D:\Temp\.arduinoIDE-unsaved20221118-6196-5kjgm6.x7t35\sweep\sweep.ino:10:29: error: call of overloaded 'setMaxSpeedDeg(int)' is ambiguous
   stepper.setMaxSpeed(100);     // скорость движения к цели
                             ^
In file included from D:\Temp\.arduinoIDE-unsaved20221118-6196-5kjgm6.x7t35\sweep\sweep.ino:3:0:
e:\arduino\libraries\GyverStepper\src/GyverStepper2.h:388:10: note: candidate: void GStepper2<_DRV, _TYPE>::setMaxSpeedDeg(int32_t) [with GS_driverType _DRV = (GS_driverType)0; GS_driverType _TYPE = (GS_driverType)3; int32_t = long int]
     void setMaxSpeedDeg(int32_t speed) {
          ^~~~~~~~~~~~~~
e:\arduino\libraries\GyverStepper\src/GyverStepper2.h:393:10: note: candidate: void GStepper2<_DRV, _TYPE>::setMaxSpeedDeg(float) [with GS_driverType _DRV = (GS_driverType)0; GS_driverType _TYPE = (GS_driverType)3]
     void setMaxSpeedDeg(float speed) {
          ^~~~~~~~~~~~~~
e:\arduino\libraries\GyverStepper\src/GyverStepper2.h: In instantiation of 'void GStepper2<_DRV, _TYPE>::setAcceleration(uint16_t) [with GS_driverType _DRV = (GS_driverType)0; GS_driverType _TYPE = (GS_driverType)3; uint16_t = unsigned int]':
e:\arduino\libraries\GyverStepper\src/GyverStepper2.h:98:24:   required from 'GStepper2<_DRV, _TYPE>::GStepper2(uint16_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t) [with GS_driverType _DRV = (GS_driverType)0; GS_driverType _TYPE = (GS_driverType)3; uint16_t = unsigned int; uint8_t = unsigned char]'
D:\Temp\.arduinoIDE-unsaved20221118-6196-5kjgm6.x7t35\sweep\sweep.ino:5:46:   required from here
e:\arduino\libraries\GyverStepper\src/GyverStepper2.h:332:35: warning: unused parameter 'acc' [-Wunused-parameter]
     void setAcceleration(uint16_t acc) {
                                   ^~~

exit status 1

Compilation error: call of overloaded 'setMaxSpeedDeg(int)' is ambiguous_

Пример скетча

```
_// крутим туда сюда, тикаем в loop
#define GS_NO_ACCEL
#include "GyverStepper2.h"
//GStepper2<STEPPER4WIRE> stepper(2048, 5, 3, 4, 2);
GStepper2<STEPPER2WIRE> stepper(3200, 9, 8, 7);

void setup() {
  Serial.begin(9600);
  stepper.enable();
  stepper.setMaxSpeedDeg(200);     // скорость движения к цели
  //stepper.setAcceleration(200); // ускорение
  stepper.setTargetDeg(360);       // цель
}

bool dir = 1;
void loop() {
  stepper.tick();   // мотор асинхронно крутится тут

  // если приехали
  if (stepper.ready()) {
    dir = !dir;   // разворачиваем
    stepper.setTargetDeg(dir * 360); // едем в другую сторону
  }

  // асинхронный вывод в порт
  static uint32_t tmr;
  if (millis() - tmr >= 30) {
    tmr = millis();
    Serial.println(stepper.pos);
  }
}_
```

GyverLibs, 2022-12-18:
Приветствую. Забавно, компилятор на этой версии перестал различать int и float?

Roman-RV78, 2022-12-18:
На 1.8.20 тоже самое

GyverLibs, 2022-12-18:
Починим, когда нибудь

GyverLibs, 2023-07-18:
пофикшено в грядущем обновлении

closed 2023-07-18 by GyverLibs