GyverLibs/GyverStepper #19: S-Curve и резонансные фильтры

temagrin, 2024-01-11:
Идея - приблизить планировщик к тому как это сделано в Klipper/Marlin но не так громозко как там и не завязано на 3d печать)
Референс: https://www.pmdcorp.com/resources/type/articles/get/s-curve-profiles-deep-dive-article
ну а фильтрами - пропускать движения на резонансных скоростях

GyverLibs, 2024-01-11:
Планировщики как раз и работают по S-curve, здесь это называется движение с ускорением, тк это физический смысл этой кривой

GyverLibs, 2024-01-11:
Нет, стоп. В статье имеется в виду jerk - ускорение третьего порядка, я не знаю как его красиво реализовать в численной модели

temagrin, 2024-01-11:
> Планировщики как раз и работают по S-curve, здесь это называется движение с ускорением, тк это физический смысл этой кривой

в текущей реализации получена трапецивидная модель движения - то есть постоянное ускорение до максимально установленной скорости и постоянное торможение. в S-Curve верно что ускорение линейно возрастает до момента ограничения скорости а термин jerk больше относится к рывку - у TinyG на графиках довольно наглядно https://github.com/synthetos/TinyG/wiki/Jerk-Controlled-Motion-Explained .
с антирезонансными фильтрами тоже кажется целочисленно не выйдет, а вот безрывковое S-curve реализовать надо подумать

GyverLibs, 2024-01-11:
Jerk это просто третья производная координаты, то есть производная ускорения
https://en.m.wikipedia.org/wiki/Fourth,_fifth,_and_sixth_derivatives_of_position