GyverLibs/GyverStepper #25: esp32 нет генерации шагов

iqubik, 2024-12-24:
У меня на esp32 gyverStepper вообще не генерирует импульсы на любых примерах из либы.
Конкретно пробовал пины для step 22 и 5. 2wire step dir config
При этом dir стабильно меняет положение соответственно текущему шагу из базового примера например.
Смотрю осциллографом - на step импульсов не завезли, а dir работает.

GyverLibs, 2024-12-24:
всё отлично работает, проверяйте  пины/подключение/скетч

closed 2024-12-24 by GyverLibs

iqubik, 2024-12-25:
Полностью переустановил arduino ide  на свежайшую 2.3.4 обновил все либы и компоненты.
Взял sweep пример и залил на классическое esp32 вообще без изменения кода скетча на 2 esp32 с использованием крайнего ядра espressif 3.1.0. Осциллограф - нет генерации step. Меняю в примере пины dir и step местами и пин который не генерировал step успешно работает с dir, а пин который ранее успешно работал с dir не создаёт импульсов для step.
ЧЯДНТ?)
Ставлю сейчас ядро версии 3.0.0

iqubik, 2024-12-25:
Ядро версии 3.0.0 есть генерация! На крайнем ядре 3.1.0 либа не даёт генерации